easyw/RF-tools-KiCAD #1: Via Fence Generator version 1.1 Error with Windows7 + KiCad 5.1.4-1

bvernoux, 2019-10-15:
I have just cloned actual repository 784871e56d02b5ddf59310450e1bc0e1ec66a7c7 to my PC **kicad5_1_4\share\kicad\scripting\plugins\RF-tools-KiCAD**
All work fine except when I click on Icon **Via Fence Generator version 1.1** which returns following error:

[image]

Details on KiCad Version used:
```
Application: Pcbnew
Version: (5.1.4)-1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.61.1 OpenSSL/1.1.1 (WinSSL) zlib/1.2.11 brotli/1.0.6 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) nghttp2/1.34.0
Platform: Windows 7 (build 7601, Service Pack 1), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.68.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.61.1
    Compiler: GCC 8.2.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

```

easyw, 2019-10-15:
Hi @bvernoux 
the error message is pointing that 'pyclipper' lib is missing... 
please have a look at the kicad forum's thread... there is a workaround to let the plugin working...
[rounded-tracks-reloaded-again-rf-tools-for-kicad](https://forum.kicad.info/t/rounded-tracks-reloaded-again-rf-tools-for-kicad/19190/57)

bvernoux, 2019-10-16:
Thanks for that amazing plugin I was waiting those RF features for years !!
Thanks for the Via Fence workaround it works perfectly on my PC (Windows7+SP1 with latest updates)

easyw, 2019-11-05:
@bvernoux 
I have included pyclipper lib on the plugin repo... could you please test the new plugins after having removed/renamed the 'pyclipper.pyd' file used as workaround?
https://forum.kicad.info/t/rounded-tracks-reloaded-again-rf-tools-for-kicad/19190/92

bvernoux, 2019-11-06:
Thanks for the final fix
I confirm I have removed the workaround and updated to latest version and all work fine (tested with Windows 7 SP1 64bits + KiCad (5.1.4)-1

closed 2019-11-06 by bvernoux