easyw/RF-tools-KiCAD #10: GetTimeStamp removed from nightlies (v5.99~v6)

raydiator2, 2020-03-08:
May not be a high priority right now since nightlies are a moving target but it appears the Get/SetTimeStamp has/is being removed so future versions of RF-tools will need a refresh
https://forum.kicad.info/t/kicad-nightly-gettimestamp-settimestamp-are-gone/21584

easyw, 2020-03-08:
@raydiator2 just FYI @maui at kicad forum is @easyw at gh :)

raydiator2, 2020-03-09:
@easyw I had a suspicion and probably should have looked at a couple of further kicad forum posts before putting in the issue. At least you already know and this can be considered public service announcement :-)
At least you know the tools are useful to others

pdp7, 2020-03-31:
I think I am running into this as well with 5.99:
```
Application: KiCad
Version: (5.99.0-1023-gd6706c257), debug build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.65.3 GnuTLS/3.6.9 zlib/1.2.11 libidn2/2.2.0 libpsl/0.20.2 (+libidn2/2.0.5) libssh/0.9.0/openssl/zlib nghttp2/1.39.2 librtmp/2.3
Platform: Linux 5.6.0-rc6+ x86_64, 64 bit, Little endian, wxGTK
Build Info:
    Build date: Mar 3 2020 11:34:43
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.67.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.65.3
    Compiler: GCC 9.2.1 with C++ ABI 1013
Build settings:
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
```
[image]
[image]

easyw, 2020-04-02:
@pdp7 @raydiator2 
I'm slightly updating the RF-tools to be kv6 compatible.
ATM I have improved the track rounder with this commit https://github.com/easyw/RF-tools-KiCAD/commit/1b3cfc6f1586b6f95129c5b53bfbd5885475a201
Not all the functions are available in kv6.. I'm investigating if this can be improved further.

pdp7, 2020-04-03:
@easyw thank you!  The track rounder is now working in 5.99
[image]

easyw, 2020-04-04:
@pdp7 
thanks for the feedback... unfortunately still there are functions missing/disabled that rely on SetTimeStamp missing in kV6