easyw/RF-tools-KiCAD #12: new branch for v6 compatibility?

matthuszagh, 2020-04-23:
I've made a few changes to my local version of this repo to get pieces working with the kicad nightlies. However, I don't want to make any PRs because I expect that would break things for v5 users. Do you think it makes sense to create a v6 branch where we can get going on breaking v6 changes?

easyw, 2020-04-23:
@matthuszagh 
it is possible to make a version working both for kv5 and kv6.
Please have a look at my latest [PR](https://github.com/easyw/RF-tools-KiCAD/commit/1b3cfc6f1586b6f95129c5b53bfbd5885475a201) to see what I did to make the track rounder working in kv6 and keep it working in kv5. Some features are lost in kv6 for the lack of kicad python API, but the plugin is still usable.
If you are going to make a PR with this approach, I will be happy to merge it.

matthuszagh, 2020-04-23:
Makes sense! [Here](https://github.com/easyw/RF-tools-KiCAD/pull/13) it is.

closed 2020-04-23 by matthuszagh