easyw/RF-tools-KiCAD #14: Trace copper pour clearance addition

matthuszagh, 2020-04-23:
I'm considering adding a plugin that could set a copper pour keepout zone for a trace. Basically, it would be very similar to how your solder mask removal feature works but for pour keepout zones. The rationale for this that I'd like to keep a large coplanar ground keepout around my microstrips so that the ground doesn't affect the characteristic impedance of the line, but setting a large clearance for the net class makes it impossible to connect the trace to pads.

I'm thinking that it might make more sense to add the feature to this repo, so that users don't have to install another plugin. Is that something you might be open to? I could also get it started in a separate repo and then we can revisit this when I have it working. Or, if you don't see the use that's fine too.

I think v6 will resolve this with the more sophisticated constraints, but in the meantime a plugin might be useful.

easyw, 2020-04-23:
@matthuszagh 
> I'm thinking that it might make more sense to add the feature to this repo, so that users don't have to install another plugin.

that sounds correct... please go ahead 😄

matthuszagh, 2020-04-24:
PR [here](https://github.com/easyw/RF-tools-KiCAD/pull/15). Closing this.

closed 2020-04-24 by matthuszagh