easyw/RF-tools-KiCAD #16: Round traces should use Arcs in kicad version >= 5.99

easyw, 2020-04-25:
recently in kv5.99 have been added round tracing format  .. track_rounder tool should use 'arcs' instead of poly lines when available
@matthuszagh would you mind to have a look at it? :Smile:

pdp7, 2020-04-25:
> recently in kv5.99 have been added round tracing format
@easyw that sounds exciting!
do you know what commits to look at?

easyw, 2020-04-25:
@pdp7 
something after this https://github.com/KiCad/kicad-source-mirror/commit/78159c67447eac6e50dd21a44191b3b32296e628
you may have a look at the feature here:
https://gitlab.com/kicad/code/kicad/-/issues/4217

pdp7, 2020-04-26:
@easyw thanks, does anything happen for you when you select 'Switch Corner Rounding'?
[image]

```
Application: Pcbnew
Version: (5.99.0-1406-gfd59660ec), debug build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.65.3 GnuTLS/3.6.9 zlib/1.2.11 libidn2/2.2.0 libpsl/0.20.2 (+libidn2/2.0.5) libssh/0.9.0/openssl/zlib nghttp2/1.39.2 librtmp/2.3
Platform: Linux 5.7.0-rc1-5.5-big-config+ x86_64, 64 bit, Little endian, wxGTK
Build Info:
    Build date: Apr 23 2020 14:33:26
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.67.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.65.3
    Compiler: GCC 9.2.1 with C++ ABI 1013

Build settings:
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
    KICAD_STDLIB_DEBUG=OFF
    KICAD_STDLIB_LIGHT_DEBUG=OFF
    KICAD_SANITIZE=OFF
```

easyw, 2020-04-26:
@pdp7 for me it is working but not very user friendly
.. anyway this should be pointed out there:
https://gitlab.com/kicad/code/kicad/-/issues/4217

easyw, 2021-09-16:
round tracks implemented with arcs for kv5.99

closed 2021-09-16 by easyw