easyw/RF-tools-KiCAD #19: Exception on python action plugin code : IndexError: list index out of range

mjm2017, 2020-09-06:
This happens with the latest's night build. (2020-09-06)
I put the files under the path (C:\Program Files\KiCad\share\kicad\scripting\plugins)
in round_trk.py  .line 149
_pcbnew_frame=[x for x in wx.GetTopLevelWindows() if x.GetTitle().lower().startswith('pcbnew')[0]

thanks.

easyw, 2020-09-17:
@mjm2017 
kicad pcbnew nightly has recently changed a lot of calls... I'm trying to check what I can do to keep my plugins running in nightly

easyw, 2020-09-17:
I've adde an issue at kicad source code:
https://gitlab.com/kicad/code/kicad/-/issues/5678

sqrtmo, 2021-01-22:
Hello, 
still nothing on this? Have the same error Kicad 5.1.9, osx 11.1.

Traceback (most recent call last):

File "/private/var/folders/9l/72_8n9zn0172rd8bbn08ypz00000gn/T/AppTranslocation/96DC57D0-3189-4184-9D04-788A137B5E2C/d/kicad.app/Contents/SharedSupport/scripting/plugins/round_tracks/round_trk.py", line 152, in Run
_pcbnew_frame = [x for x in wx.GetTopLevelWindows() if 'pcbnew' in wx._windows.Frame.GetTitle(x).lower()][0]

IndexError: list index out of range

easyw, 2021-05-12:
this issue should be closed now ... still the plugin is not fully functional for k v 5.99

closed 2021-05-12 by easyw