easyw/RF-tools-KiCAD #20: v5.99: Rounder for Tracks issue

pdp7, 2020-09-29:
I tried out the current `master` branch of this repo and the current v5.99 KiCad build.  I know that v5.99 is not officially supported but I thought I would report in case it is useful.

When I try to round a track, the plugin does create the curve out of line segments like normal but the line segments are not visible unless I click on them.  Here is a video to try to demostrate what is happening:

Feel free to close if this is not useful.  And please let me know if any more information would be useful.

```
pdp7@x1:~/.config/kicad-master/5.99/scripting/plugins/RF-tools-KiCAD$ git log --stat -1
commit d4b8b5269b974cf92a3e09e09e08e78f2f1e8c5f (HEAD -> master, origin/master, origin/HEAD)
Author: easyw <easyw2g@gmail.com>
Date:   Thu Sep 17 14:39:10 2020 +0200

    first baby steps to be v6 compatible

 round_tracks/round_trk.py | 9 +++++++--
 1 file changed, 7 insertions(+), 2 deletions(-)
```
```
pdp7@x1:~/dev/kicad/kicad$ git log -1
commit 6b958ff4e0c941893d786913353999d2e4c1985b (HEAD -> master, origin/master, origin/HEAD)
Author: jean-pierre charras <jp.charras@wanadoo.fr>
Date:   Wed Sep 23 20:58:13 2020 +0200

    Move specialized plotter classes from plotter.h to specific plotter_xx.h
    plotter.h contains now only a header common to all plotters.
    The code does not actually change, but it allows modifying a specific plotter
    without recompiling most of kicad files.
```
```
Application: Pcbnew

Version: (5.99.0-3569-g6b958ff4e), debug build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 GnuTLS/3.6.13 zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.4.0-45-generic x86_64, 64 bit, Little endian, wxGTK

Build Info:
	Date: Sep 27 2020 21:01:42
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCE: 6.9.1
	Curl: 7.68.0
	ngspice: 31
	Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=OFF
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
	KICAD_SCRIPTING_ACTION_MENU=ON
	BUILD_GITHUB_PLUGIN=ON
	KICAD_USE_OCE=ON
	KICAD_SPICE=ON
	KICAD_STDLIB_DEBUG=OFF
	KICAD_STDLIB_LIGHT_DEBUG=OFF
	KICAD_SANITIZE=OFF
```

pdp7, 2020-09-30:
Interesting observation: the tracks are rounded as expected is I save the file, close it and then re-open.
[image]

easyw, 2020-09-30:
@pdp7 
this issue is in wip... https://github.com/easyw/RF-tools-KiCAD/issues/19
Unfortunately ATM I have little time to investigate (mainly I'm using kicad stable)

pdp7, 2020-10-01:
@easyw thanks. in that case, should I close this issue?

easyw, 2020-10-01:
@pdp7 I would take this open until I will have time to fix this issue... your report has useful feedback

easyw, 2021-09-17:
@pdp7 it seems it fine now.
I'm closing it.

closed 2021-09-17 by easyw