easyw/RF-tools-KiCAD #25: Create a branch for kicad-nightly

tkuester, 2021-06-25:
Hey! Thanks for this wonderful plugin. It's a huge time saver.

I recently made the switch to nightly, and urgently needed the plugin. I started to port it over, and had to make a few ugly hacks. It's definitely not something I'd say "push out to users" yet, but it worked well enough to get the job done. I think the code can definitely be made to run on both 5.1 and 5.99 -- but it will be some effort.

Anyways! My code is in [my fork](https://github.com/tkuester/RF-tools-KiCAD/tree/kicad-nightly). If you create a separate branch, I can offer a pull request to help get things started?

easyw, 2021-09-16:
@tkuester 
sorry for the delay in reviewing your code...
I had some spare time on holidays to partially port the code to 5.99, keeping all in one single repo.
I'm going to have a look at your repo to check if your porting can be merged to the actual code status.
Thanks 
Maurice

tkuester, 2021-09-16:
No worries at all. This repo was greatly useful for a project I was working
on! I just wanted to contribute back. :)

I tried to keep my commits small and document the steps. I hope you find it
helpful!

easyw, 2021-10-14:
Most of the tools are now compatible with k 5.1 and 5.99

closed 2021-10-14 by easyw