easyw/RF-tools-KiCAD #26: Round Tracks

HaydenHu, 2021-08-04:
Can you add the function of adding arcs to other layers of graphic lines to the Round Tracks plug-in? For example, the Edge.Cuts.

luczjanoo, 2021-08-19:
You can do that by ARC icon in Kicad tools.

[image]

HaydenHu, 2021-09-28:
I mean like your plug-in, it's convenient to choose two segments at right angles and turn them into fillets!

HaydenHu, 2021-11-17:
Now that the knightly version of kicad has supported turning a 45-degree angle into an arc, your arc plug-in will not do much more, I hope you can take my advice and change the insert that draws the arc to other coatings, such as a rectangle drawn using the "edge.cut" layer, any two edges on the selected rectangle can automatically turn the right angle into a 90-degree arc, and the selected rectangle can turn all four corners into an arc, which is convenient when drawing the board.

easyw, 2023-03-20:
fillet trace or drawing is now included in stable kicad release

closed 2023-03-20 by easyw