easyw/RF-tools-KiCAD #27: Python error in Solder Mask Expander.

luczjanoo, 2021-08-19:
Hi.

If I want to start Solder Mask Expander I got error like on screen.

[image]

How to fix it ?

albtronics, 2021-08-21:
I also have the same issue.
Please help.

easyw, 2021-09-16:
Hi, I'm porting the code to k 5.99 and this tool is broken atm for k5.1.
I'm going to see if I can maintain a single repo for both of releases as I did for rf fp wizards, round tracks and tracks length.

easyw, 2021-09-17:
@luczjanoo @albtronics 
it should now be fixed
https://github.com/easyw/RF-tools-KiCAD/commit/d5a5dfef084cbfa374c06784e872ab27754ebdbb

albtronics, 2021-09-17:
Thank You so much. :)

closed 2021-10-14 by easyw