easyw/RF-tools-KiCAD #31: TraceClearance

HaydenHu, 2021-10-10:
[image]

Application: KiCad PCB 编辑器 (64-bit)

Version: (5.99.0-12741-g2662433374), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Oct  9 2021 12:00:59
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

easyw, 2021-10-10:
@HaydenHu 
I think this is an issue in kicad pcbnew.py API
the function Zone_AddPoligon hasn't changed as definition, but it doesn't work in 5.99
I think we should rise an issue at kicad source code

easyw, 2021-11-16:
It is fine now.

closed 2021-11-16 by easyw