easyw/RF-tools-KiCAD #34: KiCad 6

arnobert, 2022-01-11:
Does not work 'out of the box' in new release 6: Rounded tracks and solder mask clearance tracks cannot be deleted. KiCad crashes on closing PCB editor.

CGrassin, 2022-02-20:
To add to this issue, those are the problems I encountered with this plugin in Kicad 6 (Version: 6.0.2):
* Graphical issue: after using the **"Solder mask expander"** tool, if you delete the generated zone, it still shows up in the editor. However, it is indeed deleted and does not appear in the renders. Reloading the GUI (for instance, quitting and relaunching the PCB editor) fixes the problem. This issue does not appear when manually creating a solder expansion zone using Kicad standard tools.
* Graphical issue: exactly the same problem with the **"Trace clearance generator"**.
* Crash: the **rounded track tool** crashes the PCB editor. It is however still somewhat usable as the project can be saved, and the rounded track do not seem to crash the editor after restarting it; only immediately after using it.

This plugin is fortunately usable in Kicad 6 despite these problems, but a fix would be appreciated.

ModuloFS, 2022-02-21:
Would be very nice if these tools are available in the Plugin and Content Manager as well! Is there any plans to do so?

QGB, 2022-11-06:
kicad 6.0.9 python 3.914