easyw/RF-tools-KiCAD #35: missing metadata.json

crteensy, 2022-11-21:
I tried to add RF Tools to KiCad 6.0.4, but it seems that the plugin can't be installed because metadata.json is missing. I tried to install

- through the plugin manager and
- also by unzipping in ~/.local/share/kicad/6.0/scripting/plugins and refreshing in PCBNew -> Tools -> External Plugins -> Refresh Plugins.

Others apparently don't have this problem (maybe from an older KiCad installation?).

mini-ninja-64, 2022-11-24:
I am using KiCad 6.0.9 and I have it installed by cloning the repo into my `~/.local/share/kicad/6.0/scripting/plugins/RF-tools-KiCAD` path.

crteensy, 2022-11-24:
I had more luck after reinstalling KiCad, now also 6.0.9. Something must have been odd with my old installation.

closed 2023-03-02 by easyw