easyw/RF-tools-KiCAD #45: Rounder for tracks 2.6 not working well in kicad 7.0.0

eakhub, 2023-03-03:
Hello,
I have a problem with Rounder for tracks. If I try to round tracks going right - up, its rounded in wrong way. But it's working ok when track is going to right-down. 
Kicad 7.0.0
Rounder 2.6

See image please:
[image]
Best regards

easyw, 2023-03-04:
@eakhub 
would you mind to post a failing board?

eakhub, 2023-03-04:
@easyw 
Ok, great. Didn't know about wire fillet in K7m thanks.
Btw, I often need to make RF tracks of equal length (phase matching) but going over different trajectories. Today I should play with radius of bending to make length more or less similar. Is it possible to combine your "length measuring" tool with bending tool in an interactive manner, so that it shows the length of the track at given bending radius, or even better, calculates bending radius given the length of the track ? In this case your bending tool will not be obsolete.

easyw, 2023-03-07:
it should be now fixed in version 2.7 https://github.com/easyw/RF-tools-KiCAD/commit/575d1e54a6bba2ed981bd6f81bcd560beebc4a53

> PPS. And just want to say thank you, your RF tool package is mandatory. In our company we are able to make boards working up to 50 GHz thanks to your tools.

thanks for your feedback... it would be nice to have some image of the RF boards you have done...
you can reach me in PM at kicad.info forum as maui

closed 2023-03-08 by easyw