easyw/RF-tools-KiCAD #46: Error when attempting to use Taper in KiCad 7.0.0

regicidalplutophage, 2023-03-05:
Getting this error in scripting console when attempting to taper round pad to an arc trace segment.
When doing that to a straight segment, KiCad and scripting console both hang.
In both cases malformed zone is added.
```
9:05:04 AM: trackAngle=135.0
9:05:04 AM:  m2 sx=1.7 sy=1.7 nsx=1.7 nsy=1.7
9:05:04 AM:  m2 mod angle=0.0
9:05:04 AM:  t1 sx=1.7 sy=1.7 nsx=1.7 nsy=1.7
9:05:04 AM:  t1 track angle=135.0
9:05:04 AM: targetLength=0.017
9:05:04 AM: vpercent=0.0001
9:05:04 AM: vpercent/100=1e-06
9:05:04 AM: offAngle=0.0 vec[0]=-0.7071067811865476 vec[1]=0.7071067811865476
```

regicidalplutophage, 2023-03-05:
Just got a similar result in 6.0.3.

regicidalplutophage, 2023-03-05:
Sorry, that "error" seems to be just some misc info now that I look at it.
Nightly does produce this though:
[image]

regicidalplutophage, 2023-03-05:
Kicad hanging up was misunderstanding on my part, the issue with this plugin generating weird shapes stands
[image]

easyw, 2023-03-06:
@regicidalplutophage 
taper plugin is intended for handling trace to rectangular pads, with pad smaller than track.
Moreover this plugin ATM has mainly limited functionality for near 0° or 90° footprint rotation.

For your user case simply use tear drops (already included in main sw)
https://forum.kicad.info/t/how-to-use-teardrops-in-6-99-7-0-rc1/39522/18

regicidalplutophage, 2023-03-06:
> @regicidalplutophage taper plugin is intended for handling trace to rectangular pads, with pad smaller than track. Moreover this plugin ATM has mainly limited functionality for near 0° or 90° [...]

I see. I was actually looking for a better solution than native teardrops in 7,0 or the plugin for 6.0 as each has some shortcomings.

closed 2023-03-06 by regicidalplutophage

easyw, 2023-03-06:
I'm rewriting the comment I deleted to keep this as reminder for readme file
(thanks for closing the issue):

taper plugin is intended for handling trace to rectangular pads, with pad smaller than track. Moreover this plugin ATM has mainly limited functionality for near 0° or 90° footprint rotation. [image]
 
For your user case simply use tear drops (already included in main sw) https://forum.kicad.info/t/how-to-use-teardrops-in-6-99-7-0-rc1/39522/18