easyw/RF-tools-KiCAD #47: syntaxWarning: "is not"

peter-jensen, 2023-03-15:
I get the following error message in version : v1.2.1 minor fix for round trks kv5. 71461f358921a41977c3d9302825a04fede7f9ab

/home/xxxx/.local/share/kicad/6.0/scripting/plugins/RF-tools-KiCAD/round_tracks/round_trk.py:379: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if segments is not None and dist is not 0:

Platform OS: Ubuntu 20.04.5 LTS x86_64
Python 3.8.10 (default, Nov 14 2022, 12:59:47)

peter-jensen, 2023-03-15:
Application: KiCad
Version: 6.0.11-2627ca5db0~126~ubuntu20.04.1, release build
Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.15.0-67-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11
Build Info:
	Date: Feb  1 2023 20:48:15
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCC: 7.5.2
	Curl: 7.87.0
	ngspice: 38
	Compiler: GCC 9.4.0 with C++ ABI 1013
Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

easyw, 2023-03-15:
@peter-jensen 
it should be now solved with a new release.
https://github.com/easyw/RF-tools-KiCAD/releases/tag/v1.2.2

closed 2023-03-16 by easyw