easyw/RF-tools-KiCAD #48: All plugins do not recognize the selected object

HaydenHu, 2023-03-16:
[image]
[image]
[image]

kicad7.0.1
win10

easyw, 2023-03-16:
@HaydenHu 
1) the message says "you must select two tracks only" but from the image you have selected three objects
2) ??? it is strange
3) you may try disabling 'Removing Vias violatings clearance rules' check box

Could you please add the version of your RF-tools?
And the version of your single tool (simply put the mouse over the tool to get the tooltip with the release info)

closed 2023-03-16 by easyw

Raptotype, 2023-10-18:
@easyw 

I am seeing exactly the same error on Via fence 3.1 and Rounder 3.1 for Kicad 7.0.8. on Win 11
I tried the reinstall as suggested above but this did not fix it in my case. It seems to just not register any tracks as being selected , regardless of which settings boxes are ticked.

Thank you