easyw/RF-tools-KiCAD #56: All plugins do not recognize the selected object

Raptotype, 2023-10-24:
I am seeing exactly the same error as mentioned in https://github.com/easyw/RF-tools-KiCAD/issues/48#issue-1627220819
on Via fence 3.1 and Rounder 3.1 for Kicad 7.0.8. on Win 11
I tried the reinstall as suggested above but this did not fix it in my case. It seems to just not register any tracks as being selected , regardless of which settings boxes are ticked.

Thank you

424778940z, 2023-10-31:
same issue here on 7.0.8, none of the functions recognize the selected object

lyrain2009, 2023-11-10:
I find it's the plugin KiKit cause this issue
just uninstall plugin KiKit,and restart Kicad, my problem is solved.

easyw, 2023-12-13:
@prasetekus @lyrain2009 @Raptotype @424778940z 
this should be reported at KiKit owner