easyw/RF-tools-KiCAD #60: Many tools broken for kicad 8.0

likecabezon, 2024-02-26:
The lenght measuring tools seems to be working but other relevant tools like the taper, the round track or the clearance are broken

miguelfaf10, 2024-02-27:
With the taper tool the error is:

```
AttributeError: 'BOARD_ITEM_CONTAINER' object had no attribute 'GetOrientationDegrees'
```

HaydenHu, 2024-03-12:
> With the taper tool the error is:
> 
> ```
> AttributeError: 'BOARD_ITEM_CONTAINER' object had no attribute 'GetOrientationDegrees'
> ```
I've fixed this issue, you can see my pull request

arslan437, 2024-03-20:
I am having issue with rounding the tracks
[image]

easyw, 2024-04-02:
these errors should be solved now

arslan437, 2024-04-03:
Thanks updating it

closed 2024-04-04 by easyw