easyw/RF-tools-KiCAD #72: I accidentally entered a ` (tilde) in the Distance from Intersection, and now I can't use the plugin.

beforeyouknowit, 2025-01-20:
Hi there,

I'm a long-time user of your lovely KiCAD plugin, and I've recently been trying it out with KiCAD 9.x on macOS.

Today I accidentally entered a tilde in the Distance from Intersection value, but now I can't use the plugin at all. Here's the traceback, sanitized for anonymity,

Traceback (most recent call last):

File "/Users/user/Documents/KiCad/9.0/3rdparty/plugins/com_github_easyw_kicad-rf-tools/round_tracks/round_trk.py", line 359, in Run
calc_len = float(CalcLinLenght(aParameters)) #,aParameters.m_distanceMM.Value,aParameters.m_segments)

File "/Users/user/Documents/KiCad/9.0/3rdparty/plugins/com_github_easyw_kicad-rf-tools/round_tracks/round_trk.py", line 113, in CalcLinLenght
distI = FromUnits(float(self.m_distanceMM.Value))

ValueError: could not convert string to float: '`'
I've tried uninstalling and reinstalling the plugin via KiCAD's plugin manager, to no avail. I've tried hunting through my .kicad_pro file to see if I could find a default value that was changed to the tilde, but to no avail.

Whenever I enter a different value and hit OK, the plugin doesn't perform the rest of the rounding or confirmation dialogue. It just goes back to the PCB editor window. And when I try to use the plugin, I'm shown the same traceback error again.

Is there somewhere else in my filesystem, that might otherwise show where the default or last-used value for the plugin, that I should be looking for? Or is this possibly a bug with the plugin not gracefully handling the error, i.e. not resetting back to 0 ?

I'd be happy to help beta test any possible solutions you might have.

Thanks!

mkohler99, 2025-01-26:
So I experienced this same issue, on MacOS, KiCad 8 and decided to poke around in the plugins directory and 'fix' it. Im not sure how it gets in this state, but to clear it, I simply went to the directory containing the Rounder plugin, opened 'rt_config.ini' and edited the line that said "distance = " to read "distance = 2", saved the file, and it immediately started working again. This isn't so much a fix as it is instructions for getting past the issue.

beforeyouknowit, 2025-02-08:
Thanks @mkohler99, I found the same "fix" by editing the `rt_config.ini` file, to replace the missing/null value with a 1 (i.e. 1mm) value to "reset" it and move forward again. Hopefully @easyw can find a fix for this bug! 😆