easyw/RF-tools-KiCAD #73: 在kicad9.0中使用了Taper工具报错

chen70854022, 2025-02-26:
Taper tool error occurred in kicad 9.0

[image]

Farmbot26, 2025-03-10:
I haven't been able to use this feature of the plugin at all since 9.0 dropped.

zodoczi, 2025-03-12:
I wonder guys if anyone tried the Kicad "built in" teardrop option? That makes not such a nice conncetion but a bit similar

brendanm250, 2025-04-06:
Same issue, I either get the same message as above or Kicad just quits with no message.

kodizhuk, 2025-04-17:
me too, the same issue

pr0mixe, 2025-04-26:
Same error here on Win11, KiCad 9.0.1. 
1) An unhandled exception occured. Press "Abort" to terminate the program, "Retry" to exit the program normally and "Ignore" to try to continue.  (Retry just closes the app, no warnings)
2) Hitting "Abort" produces KiCad Error: Unhandled exception class: class std::bad_optional_access  what: Bad optional access. Then the app quits.

Please fix.

MarwanOSayed, 2025-05-16:
I've fixed the taper tool for V9. You can check it out in my fork or the pull requests.

Cocoader, 2025-05-17:
> I've fixed the taper tool for V9. You can check it out in my fork or the pull requests.

I tried your pull request and the issue persists at least on macOS.

MarwanOSayed, 2025-05-17:
> > I've fixed the taper tool for V9. You can check it out in my fork or the pull requests.
> 
> I tried your pull request and the issue persists at least on macOS.

I meant the taper wizard. Will edit my answer to correct it.
I will see what could be the problem with the taper tool.
In the meantime, as a workaround, you can use the wizard in the footprint manager to manually generate the taper you want and then use it in the PCB.

Cocoader, 2025-05-17:
> > > I've fixed the taper tool for V9. You can check it out in my fork or the pull requests.
> > 
> > 
> > I tried your pull request and the issue persists at least on macOS.
> 
> I meant the taper wizard. Will edit my answer to correct it. I will see what could be the problem with the taper tool. In the meantime, as a workaround, you can use the wizard in the footprint editor to manually generate the taper you want and then use it in the PCB.

No worries about the name — we all know what’s meant.
Thanks for checking what might still be causing it on macOS. 👍 I'm more than happy to test any potential fix.
Unfortunately, the workaround isn’t an option for me since I use the tool in a similar way to what’s shown in the .gif in the README. https://raw.githubusercontent.com/easyw/RF-tools-KiCAD/refs/heads/master/resources/pcbnew-taper-plugin.gif

MarwanOSayed, 2025-05-17:
Ok I've just uploaded a quick fix on a new branch in my fork here:
https://github.com/MarwanOSayed/RF-tools-KiCAD/tree/TaperTool-Dev

IMPORTANT: For now the clearance value has to be set manually after taper zone creation.

Cocoader, 2025-05-18:
> Ok I've just uploaded a quick fix on a new branch in my fork here: https://github.com/MarwanOSayed/RF-tools-KiCAD/tree/TaperTool-Dev
> 
> IMPORTANT: For now the taper's clearance value has to be set manually after taper zone creation. Otherwise, it will remain at zero.
> 
> I'll create a pull request if I come up with a more elegant fix that doesn't require manual clearance setup.

Your quick fix works perfectly — no more crashes on macOS and works as intended. 👍
https://github.com/MarwanOSayed/RF-tools-KiCAD/commit/54444c455c25df85ad14d7f72f30e961c31716c6#diff-a6a11504dc76d81b74ea756b836d07bf0a7b5d0c4dd43b906596c3ef4a1866d5R54

I don't see any apparent issues with setting the clearance to 0. As far as I understand it the clearance defined in the net classes will be favoured in this case and override it anyway. Let's say you have a situation with different net classes and clearances around that area:

[image]

 I would rather have this predictable somewhat ugly behaviour where the net classes are applied properly than having one global clearance that might lead to an unnoticed DRC error or to unnecessary large clearances. This can easily be fixed before getting the board that might have manufacturing defects.

Just out of curiosity (_my python skills are pretty limited_) are the changes in `rf_tools_wizards/uwTaper_wizard.py`, which you did in your pull request  https://github.com/easyw/RF-tools-KiCAD/pull/76/commits/5bb2f1eb8d280e7943d820a2d99f0dd1fb3252df, still the cause of the problem, or was the real issue **only** the `z.SetLocalClearance(0)` in `taper_fz/taper.py`?

Thanks a lot — I really appreciate your contribution.

MarwanOSayed, 2025-05-18:
> > Ok I've just uploaded a quick fix on a new branch in my fork here: https://github.com/MarwanOSayed/RF-tools-KiCAD/tree/TaperTool-Dev
> > IMPORTANT: For now the taper's clearance value has to be set manually after taper zone creation. Otherwise, it will remain at zero.
> > I'll create a pull request if I come up with a more elegant fix that doesn't require manual clearance setup.
> 
> Your quick fix works perfectly — no more crashes on macOS and works as intended. 👍 [MarwanOSayed@54444c4#diff-a6a11504dc76d81b74ea756b836d07bf0a7b5d0c4dd43b906596c3ef4a1866d5R54](https://github.com/MarwanOSayed/RF-tools-KiCAD/commit/54444c455c25df85ad14d7f72f30e961c31716c6#diff-a6a11504dc76d81b74ea756b836d07bf0a7b5d0c4dd43b906596c3ef4a1866d5R54)
> 
> I don't see any apparent issues with setting the clearance to 0. As far as I understand it the clearance defined in the net classes will be favoured in this case and override it anyway. Let's say you have a situation with different net classes and clearances around that area:
> 
> [image]
> 
> I would rather have this predictable somewhat ugly behaviour where the net classes are applied properly than having one global clearance that might lead to an unnoticed DRC error or to unnecessary large clearances. This can easily be fixed before getting the board that might have manufacturing defects.
> 
> Just out of curiosity (_my python skills are pretty limited_) are the changes in `rf_tools_wizards/uwTaper_wizard.py`, which you did in your pull request [5bb2f1e](https://github.com/easyw/RF-tools-KiCAD/commit/5bb2f1eb8d280e7943d820a2d99f0dd1fb3252df), still the cause of the problem, or was the real issue **only** the `z.SetLocalClearance(0)` in `taper_fz/taper.py`?
> 
> Thanks a lot — I really appreciate your contribution.

The real issue is that there was no standard API but rather the C++ source code of KiCAD is wrapped and interfaced to python through a wrapper called SWIG. This means that each time the KiCAD's C++ source code changes, the python function calls in the plugin have to be amended as well to reflect this change. The changes can be anything between changes in functions' signatures (the parameters it accepts as input), functions' return types, or that functions have been removed altogether.
The specific function call in `taper_fz/taper.py` that caused the problem is not the same as the ones in `rf_tools_wizards/uwTaper_wizard.py`.

Cocoader, 2025-05-18:
> The real issue is that there was no standard API but rather the C++ source code of KiCAD is wrapped and interfaced to python through a wrapper called SWIG. This means that each time the KiCAD's C++ source code changes, the python function calls in the plugin have to be amended as well to reflect this change. The changes can be anything between changes in functions' signatures (the parameters it accepts as input), functions' returns and their types, or that functions have been removed altogether and some things have to be achieved in a totally new way. Hence, for each incompatibility issue, the code has to be traced to pin down which function call/s is/are causing the problem. The specific function call in `taper_fz/taper.py` that caused the problem is not the same as the ones in `rf_tools_wizards/uwTaper_wizard.py`.

Thank you for the detailed answer.

StayWavy1, 2025-07-29:
I took in @MarwanOSayed latest update but get a division by zero error. I even tried messing with track widths, placement, units, etc and had no luck either. Happens for call to SetTaper_Zone and ends at the weaken calculation below.

taper_fz/taper.py, line 151 in __ ComputeCurvedTracks
weaken = (vpercent/100.0 ........
~~~~~~~~~~~~~~~~~~~~~~~~
ZeroDivisionError: float division by zero